iNavFlight/inav-configurator #2: ble device on windows?

bk79, 2016-05-12:
I'm able to use it with ez-gui, but Not in windows with chrome configurator

digitalentity, 2016-05-13:
Please explain more. I use bluetooth to talk to my quad via Configurator no problem.

bk79, 2016-05-14:
Me too with hc-05 module but with hm-10 module(ble4.0) there is no way to
connect from the pc because Bluetooth 4.0 protocol  doesn't use a com port
so it's impossible to connect with the configurator which asks for a com
port

digitalentity, 2016-05-14:
Oh, I see. Need to order myself a ble module to look into the issue

bk79, 2016-05-14:
It has more range or at least they say so...

bk79, 2016-05-14:
Btw On ez-gui works

skaman82, 2016-05-25:
Have exactly the same problem. Thats why I am using hc-05 module to connect with my macbook and hm-10 module with my iPhone. It sucks to change the module depending on the device i want to connect it with but thats how it is, I guess. This is technically not a issue.

skaman82, 2016-06-15:
closing since this not a bug

closed 2016-06-15 by skaman82

bk79, 2016-06-15:
No it's not a bug it's a missing feature

skaman82, 2016-06-15:
It is a hardware issue (or missing future). It is not configurator's fault that the hardware not supporting required protocols.

bk79, 2016-06-16:
No it's the configurator that is not able to handle the ble protocol since
there is no com port on the ble created