iNavFlight/inav-configurator #4: Suggestion: rework the GUI/Logo a bit?

skaman82, 2016-05-17:
I could rework the colorscheme a bit to look nicer. The dark blue color is a bit heavy on the yes :). I also think the logo could use some work – and maybe not look so much alike the CF Logo. Do you call it iNAV or iNAVFLIGHT? (first one is a bit nicer - people call you project iNAV already). @digitalentity what do you think?

digitalentity, 2016-05-17:
@skaman82 You have push access to this repo - feel free to do pretty much whatever you want :smile: 
The "official" name is INAV (capital "I"). INAVFlight was chosen to keep some resemblance to Cleanflight, but that could be dropped.

digitalentity, 2016-05-17:
I'm not a UI/UX guy so any assistance with usability and UI cleanup is very appreciated and welcomed!

skaman82, 2016-05-17:
First ideas regarding the logo. Thoughts?
[image]

DzikuVx, 2016-05-17:
My favs: 1 and 4 without map pin :)

digitalentity, 2016-05-17:
I agree with @DzikuVx - 1 and 4 are my favorite

jarno83, 2016-05-18:
My 12 points goes to ........ logo nr. +1:

sppnk, 2016-05-18:
Me too, but I would prefer to have also an airplane logo :-)

theArchLadder, 2016-05-19:
Wow, very nice logos, my vote also goes to nr. 1!
I agree that it would be nice to also have a airplane in there, but perhaps it just makes it feel cluttered...?

skaman82, 2016-05-24:
see https://github.com/iNavFlight/inav-configurator/pull/8

closed 2016-05-24 by skaman82