iNavFlight/inav-configurator #6: Add 25ms delay after each line when restoring a dump in cli

theArchLadder, 2016-05-19:
Multiple users have reported issues with restoring a dump in the CLI, flight modes are sometimes not restored, or only partially. Random settings are not set, and so on. See this issue on inav:
https://github.com/iNavFlight/inav/issues/229#issuecomment-220406898

@stronnag wrote

> This is a well known feature of all CF firmware variants, the firmware cannot process mode changes fast enough, the buffer overflows, data is lost. This is why the mwptools cf-cfi (cli backup / restore tool) uses a baud rate and context aware inter-line delay. Slower than pasting into a terminal, but reliable.

He then suggested that a 25ms delay after each new line could help.
To all javascript gurus: How can this be implement so we can try it and see if that solves the problem?

theArchLadder, 2016-05-19:
Well, that was simple

https://github.com/iNavFlight/inav-configurator/blob/master/tabs/cli.js#L102
Looks like changing the 5 to whatever milliseconds you want does the trick
At least it looks slow in the CLI, so i assume it works. I guess we just need to try and restore dumps now!

@oleost @stronnag Play around with it and see if it works for you guys, i will do the same tomorrow.
I think you have to go into Chromium->settings menu->more tools->extensions and then click "reload" on iNav configurator after the change has been made...

oleost, 2016-06-15:
100ms is added. This could be closed.

closed 2016-06-15 by skaman82