iNavFlight/inav-configurator #10: Cryptic characters in CLI

skaman82, 2016-05-25:
Sometimes when I enter the CLI Tab I see some cryptic characters.
[image]

sppnk, 2016-05-25:
Me too. I usually enter through bluetooth and EZGUI, so I dont know if the problem is in EZGUI itself. With cleanflight I dont connect much.

When I have the issue, the '#' character doenst appear, and I have to reset the FC to get it again.

Ralf-W, 2016-07-31:
Me too. But I use iNAV configurator on my laptop connected via bluetooth. So it is not coming from EZGUI itself. Sometimes I get not only cryptic characters, but some readable information too:

$M>htARM;ANGLE;HORIZON;AIR MODE;HEADING LOCK;MAG;HEADFREE;HEADADJ;NAV ALTHOLD;SURFACE;BEEPER;OSD SW;BLACKBOX;d$M&gtni
Entering CLI Mode, type 'exit' to return, or 'help'

DzikuVx, 2017-01-21:
Issue localized. It happens when there are pending MSP messages somewhere in the serial link when CLI is entered. #116 should fix this problem. Closing as implementation in place

closed 2017-01-21 by DzikuVx