iNavFlight/inav-configurator #11: Change color of progress bar from green to fancy blue

theArchLadder, 2016-05-25:
After poking around and running some search scripts i finally manage to find what controls the color of the progress bars in the firmware flasher

https://github.com/iNavFlight/inav-configurator/blob/master/tabs/gps.css#L122
Yes, in the gps.css, makes sense... NOT!

Not sure what fancy blue color code to use here, perhaps @skaman82 would like to do his magic touch on this one?

theArchLadder, 2016-05-25:
These where all the green colors my script found in inav-configurator, in case this is helpful to anyone...
[image]

skaman82, 2016-05-25:
thanks, done in https://github.com/iNavFlight/inav-configurator/pull/12

skaman82, 2016-05-25:
The blue color should be #39a9dc by the way

skaman82, 2016-05-25:
This was a quick fix for the moment. I will move that styling later to a file that makes actually sense.

theArchLadder, 2016-05-25:
Yay! I have been coding inav and flashed and test flown a ton of times and every time my eyes have been bleeding because of that green progress bar, no more though! My eyes will be so happy :smile:

closed 2016-06-13 by skaman82

digitalentity, 2016-06-13:
@skaman82 thanks!