iNavFlight/inav-configurator #18: Create calibration tab file

cnHeider, 2016-06-18:
Calibration tab, makes other tabs unusable until disconnect.
[image]

DzikuVx, 2016-06-18:
This tab is not working yet. Unfortunately we know

DzikuVx, 2016-06-18:
"Calibration" tab progress should be part of #20

skaman82, 2016-07-06:
@digitalentity Any progress on this? I can help create the GUI for this if I know how you like to implement the process.

digitalentity, 2016-07-07:
@skaman82 Blank calibration page is part of #20 

I want to have 3 calibration options on this tab:
**1. Accelerometer calibration**
Here I'd like to have one button "Calibrate" and 6 pictures of UAV in different positions (like in the wiki). By default all of them would be grey. FC will tell the Configurator which position it is calibrating and Configurator should light that picture by some color when it's working and if it's successful it should change color to green.
Also I want to have 6 boxes to show or type in current calibration values for accelerometer - accelerometer zero and accelerometer gain.

**2. Level calibration**
The button "LEVEL CALIBRATION" will simply take current tilt angles and adjust board alignment in such way that angles are zeroed. Maybe duplicate pitch-roll board alignment values here as well.

**3. Compass calibration**
Just a button and 3 input boxes to show or type in Mag zero values.

TIA for your help!

skaman82, 2016-12-10:
Will prepare the HTML/CSS for this in the next days

skaman82, 2017-01-11:
@digitalentity my part is ready
[image]

closed 2018-02-22 by DzikuVx