iNavFlight/inav-configurator #33: Position hold not visible in configurator

bazyl78, 2016-07-27:
After flashing Naze32 with latest release of inav I don's see at all any navigation flight modes in configurator. Are there any limitations for Naze 32 or I can assign modes only via CLI? I was using Inav configurator and not a Cleanflight one.

digitalentity, 2016-07-28:
Do your have GPS and compass connected and detected by flight controller?
INAV won't show any navigation modes if you don't have all required sensors
available.

bazyl78, 2016-07-28:
Ok. I thought it will be visible but not activ without gps. Will connect and try.

closed 2016-07-28 by bazyl78

davidgermany, 2024-10-31:
had the same issue.
On some drones you need to plug in the battery for the GPS to work.
Just USB doesn't always power the GPS, and when it is not powered, option will not show in INAV.