iNavFlight/inav-configurator #34: Drop PID scaling

DzikuVx, 2016-07-28:
Drop PID scaling in configurator, use CLI values where possible

DzikuVx, 2016-07-28:
@digitalentity any thought here?

theArchLadder, 2016-07-28:
I agree with this, having different scaling in the GUI and CLI is just confusing on unit-less values.

digitalentity, 2016-07-29:
I agree. No need to scale anything, just use same values as in CLI to avoid confusion.

Ralf-W, 2016-08-27:
BTW "set roll_rate =20" is displayed as 200 deg/s in configurator. Please change GUI to 20 deg/s or add a decimal Point 20.0 deg/s.

digitalentity, 2016-08-27:
@Ralf-W FYI CLI displayes rate values in 10*deg/s, so `roll_rate=20` **is** 200 deg/s. It's also documented in wiki - https://github.com/iNavFlight/inav/wiki/Rate-conversion-for-INAV-1.2

Ralf-W, 2016-08-27:
@digitalentity 
Uups. Sorry, my fault.

closed 2016-08-30 by DzikuVx