iNavFlight/inav-configurator #36: Not possible to set SerialRX provider

DzikuVx, 2016-08-11:
It is not possible to set SerialRX provider with configurator. CLI works

Ralf-W, 2016-08-12:
@DzikuVx 
Same issue here yesterday. But just downloaded and installed newest version of iNAV configurator (2 days old) and SerialRx provider can again be selected successfully from GUI for my Naze32 and Spektrum sat..

Ralf-W, 2016-08-12:
@DzikuVx 
I found the same issue with my NAZE32 and Spektrum sat yesterday with iNAV 1.2-RC1. 
With iNAV 1.1 (stable): No problems.

Just downloaded and installed newest iNAV configurator (Latest commit 9e0ba66  2 days ago ) and things get even worse: iNAV 1.2-RC1 flashed successfully. Can enter any tab but Configuration. Here it says "Waiting for data" and Packet Error counter starts to increase forever.

Flashed back to iNAV 1.1 (stable) and new configurator works without problems.

DzikuVx, 2016-08-12:
So now we have at least 2 different things. yesterday I flashed flip32 with 1.2 RC1 but 'Configuration' tab was working

ienders13, 2016-08-16:
@DzikuVx 
Same here with SPRF3 and 1.2 RC1. I successfully flashed and set up 1.2 flew it etc. Made some PID changes with EZGui and came back in, checked settings looked into setting up Tiltservo and then it stopped working. I can only update things with Cleanflight GUI at the moment.

crashfrequently, 2016-08-17:
Using official release of 1.2 for SPRACINGF3 and latest published iNav configurator, got the white screen of death when clicking on CONFIGURATION and FAILSAFE tabs. All others were OK.
Also noticed that the Packet Error count continuously increments.

Reflashed the flight controller with a self compile of the latest commit (dd6f6cd today) and the Configurator works normally. The Packet Error count remains at zero.

crashfrequently, 2016-08-17:
Should also have added that serialrx provider still cannot be set using the configurator.

ienders13, 2016-08-18:
SPRF£ delux BG Clone
Windows 10
Latest Configurator from Chrome store
All working but still can't set SBus in GUI had to do it with Cleanflight. All other aspects appear to be working.

digitalentity, 2016-08-19:
@ienders13 please try 1.2-RC2 and check if the issue is resolved for you.

ienders13, 2016-08-19:
@digitalentity sorry I missed off 1.2 RC2, I'll try again tonight. I had to re-configure some elements as the RC2 flash wiped previous settings. As part of setting up AUX channels one defaulted to a switch that caused failsafe to be initialised (Radio on at time) and so I was effectively locked out in the configurator from making changes, took me a moment to understand what was happening and once I disengaged failsaife on that switch I was able to make changes again. I doubt it was caught up in that but I'll check to be double sure.

ienders13, 2016-08-21:
Just an update, serial receiver provider box in config has no effect. Save and re-boot and it reverts to previous setting. This is not happening when using Cleanflight Configurator.

Only glitch I can see.

1.2 RC2
Latest configurator from Chromestore
SPRF3 Delux BG clone

closed 2016-08-22 by digitalentity

Ralf-W, 2016-08-22:
@digitalentity 
Thank you - but issue still active...
I deleated iNAV-Configurator V1.0.0 and installed new Version 1.0.1 from the Google WebStore today.
App shows up as 1.0.1. OK. Reflashed Naze32 with iNAV 1.1 and full chip erase. OK. Defined port for "Rx Serial" (Save) OK and configured "Rx serial" plus "Spektrum2048". Save. Log says
2016-08-22 @ 22:03:16 -- EEPROM saved
2016-08-22 @ 22:03:16 -- Device - Rebooting
2016-08-22 @ 22:03:18 -- Device - Ready
But CLI still shows "serialrx_provider = SPEK1024" and "Receiver" tab shows wrong behavior. Using CLI "set serialrx_provider = SPEK2048" makes receiver tab work as expected.
Thus: Configurator confirms "Yes, I changed serialrc_provider to SPEKTRUM2048" - but it did not happen. EEPROM was not saved correctly.

Now flashed iNAV 1.2 RC1. Success.
2016-08-22 @ 22:12:47 -- Flight controller info, identifier: INAV, version: 1.2.0
2016-08-22 @ 22:12:47 -- Running firmware released on: Jul 27 2016 12:40:31
2016-08-22 @ 22:12:47 -- Board: AFNA, version: 2
UART2 can be set to "Serial Rx". OK.
But "Configuration" tab freezes "Waiting for data"

Sorry - I cannot help you with Java at all.

Ralf-W, 2016-08-23:
@digitalentity 
Today I flashed iNAV 1.2 RC2. Success.
"Configuration" tab works.
"serialrx_provider" can be changed from GUI and is saved to EEPROM correctly. 
Issue is solved for me now.
THANK YOU

commit 43118c9f (2016-08-22): Fix SerialRX provider issue. Closes #36