iNavFlight/inav-configurator #37: Latest configurator not working with SPRF3

ienders13, 2016-08-16:
Works on cleanflight.

ienders13, 2016-08-16:
This  could be merged into #36 

Also running 1.2 RC1

DzikuVx, 2016-08-16:
I was using it on SPRF3 last weekend, no problem, so please be more specific, what is not working, which versions of configurator and firmware, operating system

feakesp, 2016-08-16:
Hi can I add that I have tested iNAV configurator on both a NAZE32 and a SP racing Evo and get the same results
Both cases I am using 1.2 RC firmware and the latest version of INAV configurator downloaded today from Chrome Store
In both cases I cannot open either the Configuration or Failsafe tabs ... I get 'waiting for data' and the packet errors begin to increment..
I have the same issue on both Windows and Mac OS

digitalentity, 2016-08-17:
There was a change that is not part of RC1. Stay tuned, RC2 will arrive
shortly.

ienders13, 2016-08-17:
@DzikuVx 
Digitalentity may have the solution but just to keep up with your question

SPRF3 delux BG clone
latest configurator from Chrome store and Self compiled from Github, both same result
1.2 RC1 Both by using configurator loading online and from downloading SPRF3 Hex off Github
Will open port and show front page but will not let me open Configuration tab - Waiting for data
I only sawe packet error in the thousands, didn't see it build up.

Hopefully the fix is already in hand.

Would this issue also affect the receiver tab, I may have just had a receiver failure coincide by accident but I lost that too.

DzikuVx, 2016-08-17:
@ienders13 I was using SPRF3 BG acro clone with 1.2RC1 and configurator and had no such issues

ienders13, 2016-08-17:
@DzikuVx 
The weird thing is, I set this all up, and flew it Configurator all working. Came back after PID tuning with Ezgui in the field and hooked it up, was working then suddenly stopped working. So I was almost thinking maybe chrome updated causing a conflict.

DzikuVx, 2016-08-17:
@ienders13 On Mac, Linux or Windows? From time to time configurator or rather serial stops working on my Mac. Sometimes Chrome restart helps, sometimes full restart is required

ienders13, 2016-08-17:
Ah, forgot about OS, Windows 10.

DzikuVx, 2016-08-17:
@ienders13 is this still happening or configurator is operationl. If this appears, is full restart working

GuinnessDublin, 2016-08-18:
I have excatly same problem. I cannot open either the Configuration or Failsafe tabs.

Using naze32 10dof, xubuntu 64bit linux, or chromebook. Configuration tab never opens. But if i use latest stable version all tabs work out of box.

DzikuVx, 2016-08-18:
Yes, we have confirmed case of those tabs stopped working in iNavFlight/Inav#470

ienders13, 2016-08-18:
All working except #36 

I'm closing this one.

closed 2016-08-18 by ienders13