iNavFlight/inav-configurator #45: INAV - Configurator reports old version

teckel12, 2016-09-29:
When running INAV Configurator v1.0.2 it reports:

"You are using an old version of INAV - Configurator. There may be a more recent version with improvements and fixes."

Yet, Google extension store shows v1.0.2 as the latest. Is there a newer version or is the app incorrectly reporting that it's old when it's not?

DzikuVx, 2016-11-24:
This issue we will have to solve in a completely different way that it is done currently

digitalentity, 2016-11-24:
Yes. We just need automatic way to figure out the date of release.

DzikuVx, 2016-11-24:
Well.. `https://github.com/iNavFlight/inav-configurator/blob/master/manifest.json`after configurator will switch to `master` `development` approach.
BTW, I'm reopening this to keep track of it

closed 2017-12-12 by teckel12