iNavFlight/inav-configurator #52: Airspeed sensor icon

digitalentity, 2016-10-17:
With upcoming airspeed sensor support we'll also need an airspeed sensor icon for Configurator header line. @skaman82 can you create one?

skaman82, 2016-12-08:
@digitalentity the airspeed sensor is basically a tube, so it is hard to make a icon out of it so the user would instantly understand what it stands for. I cam up with this. What do you think?
[image]

digitalentity, 2016-12-08:
Looks ok to me. Thanks!

<https://cloud.githubusercontent.com/assets/11768024/21031724/4c17d288-bda5-11e6-8ce6-2b9413c10595.png>

skaman82, 2016-12-08:
Great, I will add this to the configurator then - together with the red icons

digitalentity, 2016-12-09:
Great! Thanks!

skaman82, 2016-12-09:
@digitalentity it is taken care of in https://github.com/iNavFlight/inav-configurator/commit/544be1746eccf3ef58dd169c4f33df4749619468 and https://github.com/iNavFlight/inav-configurator/commit/584b8df22b9758aa1190b2b546ddbfde5c4f198a

skaman82, 2016-12-09:
just ad the class .error to the class of the specific icon (like .sonaricon > add .error) and it will be shown in red.

digitalentity, 2016-12-11:
Great, thanks!

closed 2016-12-11 by digitalentity