iNavFlight/inav-configurator #76: Issue with using Airbot with iNav, wont connect after flash.

oleost, 2016-12-06:
Hi, having issue with using iNav after flash.
When if first got the Airbot F4 with MPU6000 I manage after a while getting to work with iNav, I tried changing some gyro sync settings and after that I cannot get it working again.

I tried flashing iNav 1.3, I tried flashing iNav 1.4 and Betaflight.

When flashing betaflight it just works, no need to install Zadig driver, its there already and I dont have multiple interfaces to install driver to. (interface 0 and interface 1) which shows up dual in device manager after I installed driver on both aswell.

With iNav I tried replacing driver, removing all driver etc etc.
iNav configurator is official from google store.

First I uninstalled all driver to start over:
[image]

After that it did not work and I tried to replace interface 1 with WINUSB
[image]

That did not work either so I tried to replace interface 0 aswell, this time I have now com ports to connect to.
[image]
[image]

Flashed betaflight revo, everything is working without doing anything
[image]

And then I tried with iNav Revo target, that did not make any difference.
[image]
[image]
[image]

oleost, 2016-12-06:
Not sure if this is the right place for the issue, if it should be at main github.

Also tried flashing iNav with betaflight configurator, did not help. Also tried flashing latest 1.5 development.

There is something wierd here, dunno if its only my setup or something else.

Also I cannot flash "full erase" since then it hangs on erasing, this is apperntly a known issue since the chip is locked. https://www.rcgroups.com/forums/showpost.php?p=35965420&postcount=384

I could try and unlock it but I would like to find out the issue first.

DzikuVx, 2016-12-07:
I'd say it is somehow connected to driver. I remember I had to restore Windows to previous recovery point when I started to play whit F4 boards few times.

oleost, 2016-12-07:
Hmm, wierd. But any reason to identify it different than just Revo ? 

Anyway will try and unlock the board. 

But still finds it wierd that when flashed with iNav it shows to different interfaces and only one when betaflight.

DzikuVx, 2016-12-07:
Well, this is not REVO board. Motor 6 connected to different CPU pin, different sensors and so on... 
I think that VCP is not visible because wrong driver is loaded, not because config is wrong. After flashing with different firmware, all config is gone

oleost, 2016-12-07:
Closing this one, probably user error. (Replaced wrong driver with zadig)

Will update here if I get it working without restoring windows / reinstalling Windows

closed 2016-12-07 by oleost

oleost, 2016-12-07:
Complete reinstall including driver packages from device manager, and then installed https://quadmeup.com/stm32-virtualcomport-driver-for-windows/ this driver did the trick for me.