iNavFlight/inav-configurator #78: Idea: use sonar for obstacle avoidance

skaman82, 2016-12-10:
Just an idea: Could it be possible to use sonar for obstacle avoidance? Maybe place it at the front (facing forward) or at the top (facing to the top). Maybe even multiple modules if possible. Then configure the position in the configurator to match the orientation. What do you think?

digitalentity, 2016-12-12:
Several reasons actually:
1. Sonar modules don't like vibration
2. Sonar modules interfere with each other - it's hard to make several to
work together nicely
3. Sonar is slow to respond - 100ms is probably as good as you will get

I'm considering obstacle avoidance, but it definitely won't be for general
purpose boards.

DzikuVx, 2018-06-14:
This is firmware ( https://github.com/iNavFlight/inav ) , not Configurator topic. To avoid confusion I'm closing this one. Thanks.

closed 2018-06-14 by DzikuVx