iNavFlight/inav-configurator #90: Configurator PID tuining incompatible with latest development

digitalentity, 2017-01-03:
R/P/Y rates, TPA breakpoint is all messed up when using latest configurator master and latest development.

[image]

digitalentity, 2017-01-03:
Looks like MSP_RC_TUNING is broken for some reason

DzikuVx, 2017-01-03:
I will take a look and fix

DzikuVx, 2017-01-03:
@digitalentity I do not have an FC on hand, but seems like code did not changed around MSP_RC_TUNING frames both configurator and firmware. Could you test after flashing with full chip erase?

digitalentity, 2017-01-03:
It appears to be what @martinbudden warned about - migration to PG will break CLI. Apparently it breaks config reset and MSP as well.

closed 2017-01-03 by digitalentity

martinbudden, 2017-01-03:
Yes, there will be a few things broken for the next week or two while migration to parameter groups proceeds. I am aiming to keep broken things to a minimum.