iNavFlight/inav-configurator #95: Under what condition, "Navigation is safe" will pass?

liutairan, 2017-01-06:
I found that if I have "POSHOLD" mode in the switches, "Navigation is safe" will fail. So what is the condition for "Navigation is safe" to pass? 

I had all other terms pass, only "Navigation is safe" fail. I had GPS, and got 3D fix but not stable. Because the thick cloud, I could only connect to 4 satellite yesterday.

Is that the problem of too less satellites or anything else?

Thanks~

DzikuVx, 2017-01-06:
You need Solid 3d fix with 6 sats

liutairan, 2017-01-06:
@DzikuVx OK, I see. Thanks~

closed 2017-01-06 by liutairan

liutairan, 2017-01-16:
@DzikuVx I tested GPS outside the building and found that even when I had connections to 6 sats, "Navigation is safe" still failed. There were some clouds but not much. So what is the possible reason for this?

DzikuVx, 2017-01-16:
POSHOLD, Waypoints or RTH modes enabled. You can not take off in GPS assisted modes

liutairan, 2017-01-16:
@DzikuVx I had POSHOLD mode as AUX4 (1300-1700), but I didn't turn it on when I was checking the arm status.

liutairan, 2017-01-16:
As long as I have POSHOLD mode assigned to one aux, it will never able to be armed and "Navigation is safe" will fail.

digitalentity, 2017-01-17:
@liutairan yes, this is the intended behavior. firmware won't allow POSHOLD to activate when disarmed and will also prevent arming for safety reasons.