iNavFlight/inav-configurator #98: Invalid value not detected in configurator for failsafe_off_delay

bzdang, 2017-01-08:
Thank-you for this configurator!
I noticed that if I set failsafe_off_delay in CLI, the allowed range is 0 - 200, but the configurator accepts values up to 2000 and the value supplied is wrapped around to zero at intervals of 256, 512, 768 ....
Examples - value of 300 in configurator writes value of 44, value of 2000 writes 208.

Version info:
2017-01-08 @ 14:51:31 -- Running - OS: Windows, Chrome: 55.0.2883.87, Configurator: 1.5.2
2017-01-08 @ 14:51:35 -- Flight controller info, identifier: INAV, version: 1.5.1
2017-01-08 @ 14:51:35 -- Running firmware released on: Jan 4 2017 13:59:21
2017-01-08 @ 14:51:35 -- Board: REF3, version: 0

DzikuVx, 2017-03-10:
With new failsafe in 1.6 this ticket is not longer valid

closed 2017-03-10 by DzikuVx

Jacky2k, 2020-02-13:
I need to heat that up again. I see the exact behavior in current 2.4.0.
And I think this can be quite dangerous, when not double checking every single value you enter into the Configurator.