iNavFlight/inav-configurator #104: Remove SONAR from supported features

DzikuVx, 2017-01-13:


liutairan, 2017-01-24:
I think sonar is very useful when the UAV is near ground in the range of the sonar, so why remove it? The reading from sonar is much accurate than baro. I know if the UAV is very close to ground, the air flow will affect a lot, but when the height is about 1.8m, the effect is not so big any more.

DzikuVx, 2017-01-24:
Starting from 1.6 sonar will be disable completely also on firmware level. At this point it is not reliable at all. It was discussed and explained several times. Bottom line is: it is dangerous to use it, it does not work like expexted, is not detected correctly on multiple targets.
It might be back (should be back) in future releases

liutairan, 2017-01-24:
So if sonar is removed, how to detect the distance to the ground or ground object if an accurate reading of that variable is needed? I saw some videos people were using sonar and iNav firmware to do ALTHOLD and they look pretty good. Do you have some suggestions for this case? Thanks~

liutairan, 2017-01-24:
For example, 
https://www.youtube.com/watch?v=StKVn09OmzI&t=1s
https://www.youtube.com/watch?v=5vwoHNjnkFY

digitalentity, 2017-01-25:
@liutairan sonar may work for some, but for most it is not working properly at the moment. Also, sonar support is not implemented properly in the code at the moment. For safety of the most we temporarily disabled it. We'll look into bringing it back in v.1.7

liutairan, 2017-01-25:
@digitalentity Thanks for this reply.

DzikuVx, 2017-01-30:
Done in #131

closed 2017-01-30 by DzikuVx

Beardyblair, 2017-04-20:
Any update on this feature? I am happy to test on multiple platforms as this feature is needed.

oleost, 2017-04-21:
From INAV 1.7rc1 release notes
Sonar is back enabled with proper hardware detection. Not recommended to be used for flight yet