iNavFlight/inav-configurator #106: GUI bug, no scrollbar for menu sidebar when in full-screen mode

csurf, 2017-01-15:
I found a small bug in the markup for the GUI. When the window is maximized on a small screen (laptop), the sidebar menu can be cut-off and some of the last options aren't available, such as the CLI. There's no way to scroll down when the window is maximized. However, if the window is resized, the sidebar goes into 'icon' mode and a small vertical scrollbar appears for reaching the lower options.

skaman82, 2017-01-15:
strange, I have a scroll bar if the menu items not fitting in the window size - even in normal mode

[image]

DzikuVx, 2017-01-15:
I'm also unable to reproduce @csurf 
Which version of Configurator are you using?

skaman82, 2017-06-08:
Closing this since there has been no feedback on this for long time

closed 2017-06-08 by skaman82