iNavFlight/inav-configurator #119: Fallback models not displayed

stronnag, 2017-01-25:
Fall back model is not displayed on machines without webgl. This is a (fairly) recent regression. Tested on 
0ab6c7b8493fc5a7bcf9ebf7dd7c334ab4f618bc on a low powered ARM chromebook running Arch Linux (Chromium 55.0.2883.87).

stronnag, 2017-01-25:
Closing. A platform issue, as fallback works on an old eeepc without webgl.

DzikuVx, 2017-01-25:
@stronnag not so fast please, I was able to reproduce. Could you please post console error log from machine that is failing? 
I was able to break model disaplat by forcing CanvasMode

stronnag, 2017-01-25:
I solved the problem on the failing machine by unchecking the chrome://settings "Use hardware acceleration when available", so it seems to be a platform problem. 

I will reenable and get a log.

DzikuVx, 2017-01-25:
This looks like a problem THREE JS library. Perhaps update will be required

stronnag, 2017-01-25:
Log with no model on weak hardware:
````
main.js:59 Libraries: jQuery - 2.1.4, d3 - 3.5.6, three.js - 72
port_handler.js:71 Selecting last used port: /dev/ttyUSB0
serial_backend.js:92 Connecting to: /dev/ttyUSB0
serial.js:123 SERIAL: Connection opened with ID: 1, Baud: 115200
three.min.js:536 THREE.WebGLRenderer 72
````
Log with fallback model on weak hardware:
````
main.js:59 Libraries: jQuery - 2.1.4, d3 - 3.5.6, three.js - 72
port_handler.js:71 Selecting last used port: /dev/ttyUSB0
serial_backend.js:92 Connecting to: /dev/ttyUSB0
serial.js:123 SERIAL: Connection opened with ID: 1, Baud: 115200
CanvasRenderer.js:37 THREE.CanvasRenderer 72
````

closed 2017-09-19 by DzikuVx