iNavFlight/inav-configurator #122: TPA threshold set to 0 after applying Preset 600mm Flying Wing

DzikuVx, 2017-01-29:
@oleost I remember correctly/

oleost, 2017-01-29:
Yes

 ```
    name: "600mm Flying Wing",
        description: "Small flying wing on multirotor racer parts<br>" +
            "<span>300g-500g weight, 3S-4S battery</span>",
        features: [
            "Adjusted gyro filtering",
            "Adjusted PIDs",
            "Adjusted rates"
        ],
        applyDefaults: ["PIDs", "INAV_PID_CONFIG", "ADVANCED_CONFIG", "RC_tuning", "PID_ADVANCED", "FILTER_CONFIG", "FC_CONFIG"],
        settings: [
            presets.elementHelper("BF_CONFIG", "mixerConfiguration", 8),
            presets.elementHelper("PIDs", 0, [15, 30, 15]),  //ROLL PIDs
            presets.elementHelper("PIDs", 1, [15, 40, 15]),  //PITCH PIDs
            presets.elementHelper("RC_tuning", "roll_rate", 400),
            presets.elementHelper("RC_tuning", "pitch_rate", 150),
            presets.elementHelper("ADVANCED_CONFIG", "gyroSync", 1),
            presets.elementHelper("INAV_PID_CONFIG", "gyroscopeLpf", 1)
        ],
        type: 'flyingwing'
    }
];
```

Looking at that preset I'm guessing it will happend to all preset which does not have an defined TPA_

```
presets.defaultValues = {
    PIDs: [
        [40, 30, 23],
        [40, 30, 23],
        [85, 45, 0],
        [50, 0, 0],
        [65, 120, 10],
        [180, 15, 100],
        [10, 5, 8],
        [20, 15, 75],
        [60, 0, 0],
        [100, 50, 10]
    ],
    INAV_PID_CONFIG: {"asynchronousMode": "0", "accelerometerTaskFrequency": 500, "attitudeTaskFrequency": 250, "magHoldRateLimit": 90, "magHoldErrorLpfFrequency": 2, "yawJumpPreventionLimit": 200, "gyroscopeLpf": "3", "accSoftLpfHz": 15},
    ADVANCED_CONFIG: {"gyroSyncDenominator": 2, "pidProcessDenom": 1, "useUnsyncedPwm": 1, "motorPwmProtocol": 0, "motorPwmRate": 400, "servoPwmRate": 50, "gyroSync": 0},
    RC_tuning: {"RC_RATE": 0, "RC_EXPO": 0, "roll_pitch_rate": 0, "roll_rate": 0, "pitch_rate": 0, "yaw_rate": 0, "dynamic_THR_PID": 0, "throttle_MID": 0, "throttle_EXPO": 0, "dynamic_THR_breakpoint": 0, "RC_YAW_EXPO": 0},
    PID_ADVANCED: {"rollPitchItermIgnoreRate": 200, "yawItermIgnoreRate": 50, "yawPLimit": 300, "axisAccelerationLimitRollPitch": 0, "axisAccelerationLimitYaw": 1000},
    FILTER_CONFIG: {"gyroSoftLpfHz": 60, "dtermLpfHz": 40, "yawLpfHz": 30, "gyroNotchHz1": 0, "gyroNotchCutoff1": 0, "dtermNotchHz": 0, "dtermNotchCutoff": 0, "gyroNotchHz2": 0, "gyroNotchCutoff2": 0},
    FC_CONFIG: {"loopTime": 2000}
};
```

oleost, 2017-01-29:
Using `diff all` I get this

```
set fw_p_pitch = 15
set fw_i_pitch = 40
set fw_ff_pitch = 15
set fw_p_roll = 15
set fw_i_roll = 30
set fw_ff_roll = 15
set fw_p_yaw = 85
set nav_fw_pos_xy_p = 65
set nav_fw_pos_xy_i = 120
set nav_fw_pos_xy_d = 10
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 0
set roll_rate = 40
set pitch_rate = 15
set yaw_rate = 2
set tpa_breakpoint = 0
```

Seems like it affecting more values to be changed from the default values of firmware.

set yaw_rate = 2
set tpa_breakpoint = 0
set nav_fw_pos_xy_p = 65
set nav_fw_pos_xy_i = 120
set nav_fw_pos_xy_d = 10
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 0

DzikuVx, 2017-01-29:
OK, I see what happened here. All values from MSP_RC_TUNING are broken
Fix is on the way

DzikuVx, 2017-01-29:
Another problem found; iNavFlight/inav#1187 that makes Presets broken ATM

oleost, 2017-01-29:
@digitalentity @DzikuVx Bench tested both new firmware and configurator together now. Seems to work as expected.

commit bb1c674e (2017-03-05): fixes #122