iNavFlight/inav-configurator #144: Rename "magnetometer" to "compass"

digitalentity, 2017-02-06:
This is merely a terminology issue, however...

>The magnetometer measures all aspects of magnetic flux including strength and XYZ of the field for the geoligists and surveyors  - however, pilots navigate the helicopter with a "compass".   The compass is an instrument corrected for deviation errors and declination by the navigator/pilot.

While our sensor is by technology a "magnetometer" we use it to get reference direction of magnetic North, using it as a "compass". I suggest renaming the "Magnetometer" to "Compass" in the Configuration tab to reflect actual usage of the sensor.

Dronek, 2017-02-20:
Rather than completely renaming it I suggest to match the name individually to the respective context.
For example: 
- calibration procedure applies to the actual magnetometer (no renaming)
- hardware selection applies to the magnetometer hardware (no renaming)
- flight mode with mag heading applies to the usage as a compass (renaming)

Even though the main functionality of the magnetometer in iNav may be covered by the name "compass" I wouldn't exclude possible exceeding functionalities of the magetometer for iNav.

oleost, 2017-02-21:
I also vote for renaming all to compass. It is an compass, which is used as compass. 

https://cdn-shop.adafruit.com/datasheets/HMC5883L_3-Axis_Digital_Compass_IC.pdf
[image]

digitalentity, 2017-02-21:
Sensor manufacturers refer to their chips as "compass" why should we name them differently?

oleost, 2017-02-21:
Exactly my point.

Dronek, 2017-02-21:
> Sensor manufacturers refer to their chips as "compass" why should we name them differently?

The chip manufacturer decided to call the chip _compass_ whereas the actual built-in sensors are of magnetoresistive type and it can be used for **compassing and magnetometry** as stated in the PDF above. It means, this sensor is capable of measuring the magnetic field strength in addition to its direction which basically exceeds the functionality of a classical compass. AMR sensors are/were also used in harddrives for example.

My suggestion was based on a fundamental viewpoint and without insisting on it I think it won't hurt anybody when you rename it to compass completely ;)

digitalentity, 2017-02-22:
Yes, that's why I'm saying that technically it's a magnetometer, but we only use it as a "compass" sensor. Also it's more user-friendly - everybody knows what "compass" is, but many people ask "what the heck is that magnetometer sensor and why do I need it"

closed 2020-01-25 by DzikuVx