iNavFlight/inav-configurator #158: Feature reqest: Create page backup setting to file and restore setting from file

Fameing, 2017-03-01:
Hi,

Add a new Page or block in setup page backup setting to file and restore setting from a file.

DzikuVx, 2017-03-01:
Fot the time being please  use `diff all` CLI command

Fameing, 2017-03-01:
But the best one button to restore the configuration, and a button to avoid updating the configuration file directly.

DzikuVx, 2017-03-01:
See #145

Fameing, 2017-03-01:
CLI command to do several operations to save the current configuration file. For beginners not immediately clear how to restore the configuration.

DzikuVx, 2017-03-01:
I'm not saying that this function will not come back. I'm just saying that old implementation was removed as not working like expected and was not really backuping and restoring.

Backup/Restore will come back if __working__ solution will be found

closed 2019-07-29 by DzikuVx