iNavFlight/inav-configurator #162: Problem: Calibrate magnetometer button greyed out

avirotstein, 2017-03-10:
Calibrate magnetometer button in the inav configurator (1.6.2) is greyed out unless first clicking sensors tab and then clicking back to the setup tab

AustinSaintAubin, 2017-04-23:
Problem remains in Configurator 1.7.0 w/ firmware ( INAV/SPRACINGF3MINI 1.7.0 Apr 19 2017 / 09:28:25 (65cbad0) )

dragon1020, 2017-06-05:
Problem remains in Configurator 1.7.1. :(

closed 2017-09-29 by DzikuVx

Ogocology, 2018-02-24:
So sorry to post on an old issue.
I have this problem at the moment and I cant figure out exactly how to make it go away.

Any help please?

cdsisf, 2018-03-31:
[image]
the same problem here, here are my connections

Ogocology, 2018-04-06:
Hello,For the calibration grey out issue, the temporary fix is to click on any other tab ( like configuration first) then go back to calibration. That solved it for me.
Your GPS showing red possibly means you got the connection wrong. Switch the TX and RX connections and check.
Br,Ogocology

cdsisf, 2018-04-06:
Changing Tab does not solve the problem. You really are right, by doing this I will activate the button but after finishing the calibration time I can not record the calibration. Maybe it's the MAG that has something wrong. The MAG appears in blue but if turned upside down it turns red. Although it seems that the MAG (HMC5983) is recognized, in the Tab of the sensors shows no sign of life.