iNavFlight/inav-configurator #167: New 5" preset and rename existing one.

oleost, 2017-03-11:
Current 5" preset applies:

set looptime = 1000
set gyro_sync = ON
set gyro_sync_denom = 4
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_notch1_hz = 400
set gyro_notch1_cutoff = 300
set gyro_notch2_hz = 200
set gyro_notch2_cutoff = 100
set motor_pwm_rate = 2000
set motor_pwm_protocol = ONESHOT125
set attitude_task_frequency = 100
set async_mode = ALL
set mc_p_pitch = 58
set mc_i_pitch = 50
set mc_d_pitch = 22
set mc_p_roll = 43
set mc_i_roll = 40
set mc_d_roll = 20
set mc_p_yaw = 70
set dterm_lpf_hz = 80
set dterm_notch_hz = 260
set dterm_notch_cutoff = 160
set roll_rate = 80
set pitch_rate = 80
set yaw_rate = 65
```

How about renaming it to be perfectly clear that this is ONLY for F3 and newer and without GPS.
So example `5" Racer without GPS. Only for F3/F4 based FC`

And then create a new preset without async mode and higher looptime.
Example

`5" quad` Preset

```
set looptime = 1500
set gyro_sync = ON
set gyro_sync_denom = 12
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_notch1_hz = 400
set gyro_notch1_cutoff = 300
set gyro_notch2_hz = 200
set gyro_notch2_cutoff = 100
set motor_pwm_rate = 2000
set motor_pwm_protocol = ONESHOT125
set mc_p_pitch = 58
set mc_i_pitch = 50
set mc_d_pitch = 22
set mc_p_roll = 43
set mc_i_roll = 40
set mc_d_roll = 20
set mc_p_yaw = 70
set dterm_lpf_hz = 80
set dterm_notch_hz = 260
set dterm_notch_cutoff = 160
set roll_rate = 80
set pitch_rate = 80
set yaw_rate = 65
```

Is there anyone with F1 based FC like Naze32, Flip32 etc that can verify that these settings works nice with navigation features?

andre92, 2017-03-15:
It won't work. The FC has cpu protection and won't set looptime lower than 2000 with gps enabled. Could we remove this protection for 1.7 and just set 2000 as the default?

oleost, 2017-03-15:
Hmm, but with 'looptime = 2000' it works fine?

andre92, 2017-03-15:
yes, that is the fastest my Naze will do on 1.6.1. Also, no async processing available.

DzikuVx, 2017-03-15:
Implementation in #171

closed 2017-03-15 by DzikuVx