iNavFlight/inav-configurator #168: Servo tab not allowing mixer source inversion

digitalentity, 2017-03-12:
We don't have access to source inversion for servo mixer inputs.

[image]

I suggest replacing the marked area (which is **channel to forward**) with mixer source inversion flags.
```
    INPUT_STABILIZED_ROLL = 0,
    INPUT_STABILIZED_PITCH,
    INPUT_STABILIZED_YAW,
    INPUT_STABILIZED_THROTTLE,
    INPUT_RC_ROLL,
    INPUT_RC_PITCH,
    INPUT_RC_YAW,
    INPUT_RC_THROTTLE,
    INPUT_RC_AUX1,
    INPUT_RC_AUX2,
    INPUT_RC_AUX3,
    INPUT_RC_AUX4,
    INPUT_GIMBAL_PITCH,
    INPUT_GIMBAL_ROLL
```

oleost, 2017-03-12:
I'm example using aux to servo to be able to trim in flight by adjusting servo midpoint with aux channel.

Also how are one able to control which aux Channel to move servo gimbal feature if you remove it?

digitalentity, 2017-03-12:
@oleost good point. Then we need something else for source inversion config - `smix reverse` in CLI

oleost, 2017-03-12:
But what do you need it for? We have servo reverse by setting -100 ?

digitalentity, 2017-03-12:
It's not servo reversing, it's servo input reversing.

oleost, 2017-03-12:
Yes it's reversing the output from mixer to servo if I understand correctly.

digitalentity, 2017-03-12:
@oleost no, it's reversing the input:

RC/Stabilisation -> `smix reverse` -> servo mixer -> `servo rate` -> servo

oleost, 2017-03-12:
Yes what I meant.

closed 2018-05-02 by DzikuVx