iNavFlight/inav-configurator #170: Disconnects and will not reconnect when saving from cli

DzikuVx, 2017-03-15:
> Currently working on some custom mixes in CLI on a late 2011 15" MacBook pro, when I save and reboot in CLI and wait for the reboot to be confirmed complete, then switch to another tab, it acts as though no board is connected, and when I attempt to reconnect it refuses until I unplug the board and restart inav completely. CTO has tested this on his HP laptop running Windows 10 and has the same issue. Causes big delays in our work, as any time we wish to save changes in CLI we have to restart the entire system. Anyone else experienced this?

> We're running a v6 naze just for added info.

DzikuVx, 2017-03-15:
@digitalentity @oleost @stronnag I'm looking at F1 reconnect problems and looks like yes: F1 does not reconnects when leaving CLI. INAV 1.2.2 that I had on board did not had this issue so I suspect this is firmware dependent after all. 

But I will try to find a fix in Configurator since it should be able to detect reboot in all conditions

stronnag, 2017-03-15:
We've probably added enough init delays and init changes since 1.2.2 to break it.

DzikuVx, 2017-03-15:
M'key... the issue is `CLI_MINIMAL_VERBOSITY` in firmware. Configurator detects "working" CLI mode by detecting string "CLI" returned by FC when entering it. 
`Entering CLI Mode, type 'exit' to return, or 'help'`
Since `diff` and `CLI_MINIMAL_VERBOSITY` were added, that string is no longer returned and Configurator is unable to detect that he is in CLI mode. So does not knows that it should handle reboot sequence. 

That gives up 2 options:

1. Short string `CLI` will be returned by F1 targets so Configurator knows what is happening
1. We loosen state validation in Configurator and assume CLI mode is entered when we just entered the tab or FC returned anything on serial.

@digitalentity @martinbudden @stronnag ?

stronnag, 2017-03-15:
betaflight at least issues a terse 'CLI'. We should probably do the same.

digitalentity, 2017-03-16:
Please test if https://github.com/iNavFlight/inav/pull/1417 fixes the issue

DzikuVx, 2017-03-16:
iNavFlight/inav#1417 solves the problem, closing

closed 2017-03-16 by DzikuVx