iNavFlight/inav-configurator #179: No GPS map displayed on GPS page

ed4353, 2017-04-01:
The problem is that the GPS map on the GPS tab will not display. It looks like it's trying to display it, but then goes blank.
I have a good 3D gps fix. 10 to 14 satellites in view outside. No errors, no timeouts. M8N GPS.
If I click on the GPS coordinates on the left side of the GPS window (home location), the map comes up in another window, outside of INav Configurator.
SP F3 EVO board.
1.6.2 Configurator. 1.6.1 Firmware.
Windows 7, 32 bit.   It's a new build with nothing installed except Windows Updates, Chrome, and Configurator.
 I also tried Win 7, 64 bit with same results. I turned off the firewalls, and tried a different network.
Everything configures OK, and everyting seems stable.
Thank you.

DzikuVx, 2017-04-01:
This will be fixed in next release. We are above free google maps quota.

ed4353, 2017-04-01:
Thank you very much.

DzikuVx, 2017-04-03:
@ed4353 new Configurator has been released, you might want to test that.
Until then, I'm closing this one

closed 2017-04-03 by DzikuVx

ed4353, 2017-04-03:
Works perfectly. Thank you.

ecj