iNavFlight/inav-configurator #205: WingWing Z-84 Preset - Add Servo Reverse

DzikuVx, 2017-06-09:
Moved from iNavFlight/inav#1746

Howdy,

In the next available release, can you set this:

servo 3 1000 2000 1500 -100 -1

As part of the preset for the WingWing Z-84 so that the servos work correctly. Right now servo 3 is backwards and setting this as part of the preset would save a step for a new user (and potential foam snow)

ta!

Matt :)

moggiex, 2017-07-02:
Howdy,

I'm not sure how far you'd like to go with presets for models, if you'd like to reference more models, here are two more:

**C1 Chaser**
- Same settings as the Z-84 preset
- `servo 3 1000 2000 1500 -100 -1` 

**Phantom FX-61 Flying Wing**
- Same settings as the Z-84 preset
- `servo 4 1000 2000 1500 -100 -1 ` 

By setting the correct servo movements as part of the preset, this will heavily reduce the number of "annoyed" pilots who either forget to check the servo movements or just don't know how to.

Also note in both instances the servo mounting holes are "pre-made" as part of the mould (like the Z-84) so unlikely to be adapted by the model owner.

Matt

fiam, 2017-08-24:
@moggiex I think the problem with servo reversing is that servo rotation direction is not standarised, so some users will need to invert servo 3 while others will have to invert servo 4. Do you know if one direction is more prevalent or is it roughly a 50/50 split?

P.S. Love your YT channel. Keep up the good work!

moggiex, 2017-08-24:
Howdy, not sure I see your concern @fiam, the receiver inputs must work in one specific direction and the servos placement is fixed on these models, so reversing them so they work correctly should be fine, Matt

fiam, 2017-08-24:
@moggiex Some servos move CW when the signal is > 1500µs, while others move CCW, depending on what the manufacturer chose. So even if the models have the servo holes already cut out the user might need to reverse the other servo, depending on how the ones they bough they rotate.

digitalentity, 2017-08-26:
I believe servo reversal has to be done based on servos used. There is no standard for servo directions - therefore we can't create a preset that will suite all setups. In such case it's safer not to change preset at all.

fiam, 2017-08-27:
Maybe we should add a servo wizard? It shouldn't be too hard to put the aircraft into passthrough mode and send e.g. PWN = 1800 to a servo and ask the user if it's moving up or down (left or right when asking about the rudder), then reverse if it's moving in the opposite direction than we're expecting.

closed 2020-03-03 by DzikuVx