iNavFlight/inav-configurator #209: GPS map is blinking too much when fix is flaky

FlorinAndrei, 2017-06-12:
When GPS has a fix, the map is shown. When the fix is lost, the map is removed and an empty rectangle is shown.

When the fix is flaky, this is very annoying. The map is blinking on and off like one of those seizure-inducing manga cartoons. Please do not remove the last known map when the fix is lost; after all, how much could the position change in a few seconds? Leave the map displayed; instead of removing it, show some other way that the fix is lost.

E.g. you could put a colored border around the map, let's say red, a few pixels wide, when the fix is lost. The map framed in red would indicate that this is a cached position and the fix is not working momentarily. Or change the color of the Current GPS Location title bar above the map - say, green for live GPS fix, red for cached position. Or use some other way to indicate that the position is not from the current fix, but it's cached.

Speaking in general, I think it's a good idea to leave the last known map on the screen regardless of what happens - just indicate clearly that it's a cached position, not an actual live GPS fix. Oftentimes, I would like to examine the map more closely even when I don't have a fix.

Thanks!

Beardyblair, 2017-06-12:
Totally agree, may work better with the position icon as green or amber if signal lost. Perhaps red when not in the air/ not armed?

DzikuVx, 2018-11-19:
in #595 map is updated only when valid fix is available, but not hidden when no fix at all. This should solve the problem

closed 2018-11-19 by DzikuVx