iNavFlight/inav-configurator #210: build fail on ubuntu 16.04

sneiman, 2017-06-24:
Hi.  Running Ubuntu 16.04 LTS, on Intel. Doing some machine vision additions to inav for a project - and want to add one option under gps in configurator. So, I cloned configurator git, installed nodejs, and npm packages. cd'ed to project folder and gave it a try, with the following result:

seth@chimp:~/dev/inav-configurator$ sudo npm install

> nw@0.18.8 postinstall /home/seth/dev/inav-configurator/node_modules/nw
> node scripts/install.js

sh: 1: node: not found
npm ERR! Linux 4.4.0-79-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! node v4.2.6
npm ERR! npm  v3.5.2
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn

npm ERR! nw@0.18.8 postinstall: `node scripts/install.js`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the nw@0.18.8 postinstall script 'node scripts/install.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the nw package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node scripts/install.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs nw
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls nw
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/seth/dev/inav-configurator/npm-debug.log

What am I missing?

Thanks, 
seth

DzikuVx, 2017-06-24:
This is Ubuntu problem known for years. see here https://stackoverflow.com/questions/18130164/nodejs-vs-node-on-ubuntu-12-04

closed 2017-06-24 by DzikuVx

sneiman, 2017-06-24:
Thank you!

Sent from mobile device - sorry for typos and brevity.

This entire message is confidential. If it isn't intended for you, you may not use it - so please throw it away and forget about it.

aogrcs, 2019-05-13:
Dear, @sneiman 
Have you already successfully built inav-configurator on Ubuntu 16.04, thanks

sneiman, 2019-05-13:
I believe I did. I cannot check, unfortunately, because I am now on 18.04.

aogrcs, 2019-05-13:
Dear @sneiman Do you mean that you used 18.04 to build inav 2.0.0 and inav-configurator 2.0.0? Thanks
Could you give more detail about how to build them?
Sincerely

sneiman, 2019-05-13:
I last rebuilt inav-configurator on Dec 2, 2018. This was on Ubuntu 18.04.

 

I followed the instruction from the github repo:

 

https://github.com/iNavFlight/inav-configurator

 

I used the instruction for ‘Installation’ on ChromeOS on Linux. They are some issues with nodejs module names getting node installed – but these are well known on Linux, and you can google node on ubuntu to resolve them.

 

Good luck,

 

s

aogrcs, 2019-05-14:
Hi @sneiman Ubuntu 18.04 worked as the instruction, thanks

sneiman, 2019-05-14:
glad it worked out

Sent from mobile device - sorry for typos and brevity.

This entire message is confidential. If it isn't intended for you, you may not use it - so please throw it away and forget about it.