iNavFlight/inav-configurator #213: Motor Test Grayed-Out When 3D Mode Is Enabled.

KenImhof, 2017-07-02:
Motor Test Mode is grayed-out and cannot be selected to control the motors when 3D mode is enabled.  It should work like Betaflight with the default motor slider position being a the 3D neutral (1500MS).  This feature is necessary to test the motors and calibrate the ESCs and is very much needed even with 3D setups.

shellixyz, 2019-03-24:
3D mode will be removed soon replaced with better RC input and mixer functionality

closed 2019-03-24 by shellixyz

KenImhof, 2019-03-25:
There is an every growing group of us flying iNav on 3D reversing quads, so we hope that this great feature is not removed.  No changes are needed in the base code.  I have also incorporated may features on-top of iNav for 3D quads such as:

 

•        Compatibility and smoother GPS HOLD operations with 3D reversing propeller quadcopters

•        Automatic 3D arming when the low throttle position is reached for an easy arming sequence

•        Incorporates Hard Deck and Rescue Modes

•        Incorporates automatic RTH when warning battery voltage is reached for a 10-second duration

•        Improved 3d motor reversals

•        Incorporates 3d heading hold mode

•        Incorporated 3d Inverted Position Hold.  Activate when inverted by toggling the Hard Deck1 Switch ON/OFF/ON within two seconds.

•        RTH without rescue when QUAD is safe (above 1000cm and downward velocity less 200 cm/s). Low battery RTH always includes rescue punch-out for visible low battery RTH operation

•        Updated heading hold mode for compatibility with inverted and high title angle orientations

•        Improved normal and inverted navigation (simulator mode) similar to regular flight (reduced expo) but with GPS stabilization

 

I am also looking into fixing dshot to work with 3d reversing motor similar to BetaFlight.  At present, ESCs disarm and motors frequently stop in mid-flight.

 

For more information in iNav3d, please refer to:

https://github.com/KenImhof/iNav3d/releases/tag/2.1

shellixyz, 2019-03-25:
Well the possibility is not removed. It is just that the RC inputs/outputs are being reworked so it won't be necessary to have a specific "3D feature".