iNavFlight/inav-configurator #219: OmnibusF4SD

ThomasTreder, 2017-07-30:
Hi , i owned a Banggood F4 Pro V3  ( https://www.banggood.com/30_5x30_5mm-Betaflight-Omnibus-F4-Pro-V3-Flight-Controller-Built-in-OSD-Barometer-SD-Blackbox-p-1163085.html?rmmds=myorder) 
on Betaflight the Bord is Called OmnibusF4SD, but INAV do not have such a board. is there any Board i could choose to flash  Inav on it.
Becourse the layout of this board is perfect for my incoming wing ^^

Greetings Thomas

DzikuVx, 2017-07-30:
Omnibus F4 V3 is what you are looking for

ThomasTreder, 2017-07-30:
mmh this also does not work... it say`s erasing, but after a few seconds  the bar on top only says 
"2017-07-30 @ 20:07:12 -- USB device successfully closed"

DzikuVx, 2017-07-30:
Do you have "Full chip erase" enabled? If so, disable it

closed 2017-09-08 by DzikuVx