iNavFlight/inav-configurator #238: npm install fails

DzikuVx, 2017-09-07:
```
npm ERR! Error: No compatible version found: temp@'github:adam-lynch/node-temp#remove_tmpdir_dep'
npm ERR! Valid install targets:
npm ERR! ["0.2.0","0.3.0","0.4.0","0.5.0","0.5.1","0.6.0","0.7.0","0.8.0","0.8.1","0.8.2","0.8.3"]
npm ERR!     at installTargetsError (/usr/share/npm/lib/cache.js:719:10)
npm ERR!     at /usr/share/npm/lib/cache.js:638:10
npm ERR!     at saved (/usr/share/npm/node_modules/npm-registry-client/lib/get.js:142:7)
npm ERR!     at /usr/lib/nodejs/graceful-fs/polyfills.js:133:7
npm ERR!     at Object.oncomplete (fs.js:107:15)
npm ERR! If you need help, you may report this log at:
npm ERR!     <http://github.com/isaacs/npm/issues>
npm ERR! or email it to:
npm ERR!     <npm-@googlegroups.com>

npm ERR! System Linux 4.4.0-43-Microsoft
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "update"
npm ERR! cwd /mnt/c/Users/pspyc/Documents/Projects/inav-configurator
npm ERR! node -v v0.10.25
npm ERR! npm -v 1.3.10
```

Found by @digitalentity confirmed by my on Linux subsystem of Windows 10.

I suspect bug in one of the dependencies. @fiam ideas?

fiam, 2017-09-07:
@DzikuVx That package is a direct dep of nw-builder, according to its [package.json](https://github.com/nwjs-community/nw-builder/blob/master/package.json). npm seems to be able to resolve it for me. Maybe you're running and older version of npm? I'm running 5.3.0.

fiam, 2017-09-07:
@DzikuVx Just noticed that that you're running npm-1.3.10. Are you able to update or is it the latest version available for the Linux subsystem on Windows?

DzikuVx, 2017-09-08:
update to new npm/node solved the problem

closed 2017-09-08 by DzikuVx