iNavFlight/inav-configurator #243: RTC and TZ support (discussion)

fiam, 2017-09-12:
Now that we have RTC and TZ support, we need to decide how we want to support those in the configurator.

Regarding the TZ, I'd go with an explicit dialog asking the user if he wants to change the TZ when the host (e.g. the computer running the configurator) has a different TZ than the FC.

Regarding the RTC, I'm not sure what we should do. Automatically setting the time might cause issues for users running the configurator before takeoff (i.e. incorrectly set time).

Please, share your opinions on this one.

digitalentity, 2017-09-13:
>Regarding the TZ, I'd go with an explicit dialog asking the user if he wants to change the TZ when the host (e.g. the computer running the configurator) has a different TZ than the FC.

Sounds good. Would also be good to have "don't ask me again" checkbox.

> Regarding the RTC, I'm not sure what we should do. Automatically setting the time might cause issues for users running the configurator before takeoff (i.e. incorrectly set time).

How about getting the time from FC - if it's valid (set by GPS) - don't do anything. If it's not set yet - ask user if configurator should set the RTC (with option to opt out).

Obviously we need to have configurator local settings to change this behavior.

MrD-RC, 2022-11-26:
TZ offset is now in configurator.

closed 2022-11-26 by MrD-RC