iNavFlight/inav-configurator #278: Require the configurator version to be >= than the INAV version

fiam, 2017-10-11:
Due to the significant changes we've been introducing in the format of several messages when introducing new features, I think we should add a mechanism to check that the configurator version is greater or equal than the INAV version being configured. Otherwise we're risking corrupting some settings or not setting them at all.

For example, configurator 1.7.3 will happily manipulate the feature bits to set the RX in 1.7.4 (rather than using the receiver configurator message) and it will incorrectly tell the user the settings have been saved.

Best way to future proof this is to fallback to CLI when configurator version < INAV version. Additionally, we could show a banner at the top or the CLI with the link to the configurator releases page.

Linjieqiang, 2017-10-14:
That sounds great..

DzikuVx, 2019-05-24:
done in #757

closed 2019-05-24 by DzikuVx