iNavFlight/inav-configurator #287: 1.8.0 not able to set Receiver Mode of 1.7.3 target AIR3 ; backward compatibility issue

AlouetteIII, 2017-10-18:
If i use 1.8.0 to change the receiver mode of a board coded 1.7.3 ; for example to Multiwire - on reboot it shows "no receiver" ; or other settings,  similar - whatever is chosen / saved - after beboot is corrupted to another    = bug

if i use 1.7.3 to change the receiver mode of a board coded 1.7.3 ; for example to Multiwire - on reboot it shows multiwire - correctly saving and showing the new mode.

digitalentity, 2017-10-18:
@AlouetteIII please make sure you are using latest Configurator (1.8.0). We've changed the receiver code considerably, you need to use latest Configurator to be able to configure it.

fiam, 2017-10-18:
I can reproduce this one. The configurator actually sets `feature RX_PPM`, but doesn't recognise it properly. Working on a fix.

digitalentity, 2017-10-18:
Thanks @fiam

PR #289 (2017-10-18): Don't test for RX_NONE on INAV <= 1.7.3
Since in RX_NONE bit is undefined, bit_check() will always return
true. Fix is just removing RX_NONE from the available RX types
on versions using feature bits for RX, since they won't allow
not having a RX configured (FC will fall back to RX_PPM).

Fixes #287
commit e3821b8a (2017-10-18): Don't test for RX_NONE on INAV <= 1.7.3

Since in RX_NONE bit is undefined, bit_check() will always return
true. Fix is just removing RX_NONE from the available RX types
on versions using feature bits for RX, since they won't allow
not having a RX configured (FC will fall back to RX_PPM).

Fixes #287
closed 2017-10-18 by digitalentity