iNavFlight/inav-configurator #288: more buadrate options for telemetry

metalnow, 2017-10-18:
The maximum buadrate of telemetry is 115200. It should be increased to 921600 to match firmware.
[image]

digitalentity, 2017-10-18:
Do you have any usecase for telemetry baud rate to be higher than 115200?

metalnow, 2017-10-18:
I use raspberry pi connecting to SPRF3 through mavlink protocol in 921600 baudrate. I manually changed baudrate in SPRF3.

digitalentity, 2017-10-18:
MAVLink works perfectly fine at 19200 baud. Why you need almost 1 megabit?

metalnow, 2017-10-18:
1. push to the limit. I want to try to send/receiver more data from rapsPi.
2. sometime will get package lost under 115200 ( don't know why yet, have to debug ). 921600 got more stable.

closed 2025-01-19 by b14ckyy