iNavFlight/inav-configurator #292: Dataflash downloading hangs on MAC

krzysztofmatula, 2017-10-18:
Since MSPv2 was introduced I'm unable to perform data flash downloading from the iNav configurator.
Always at some point the download hangs and the dialog can't be canceled.

I see 2 problems here:

1. It happens that bigger data chunks can be received corrupted from serial on Chrome on MAC. AFAIK the issue was analysed by betaflight-configurator developers and the conclusion was that the problem is on the Chrome side. They now rely on crc-checking and retransmissions and everyone is happy.

2. iNav-configurator implements retransmission on packet crc error, but it seems it not always work. First, after retryCounter (set to 5) is reached, the transmission hangs. Simple increasing the retry limit didn't work for me. Soon, bigger number of corrupted packets happened in a row and the code hanged anyway...

krzysztofmatula, 2017-10-18:
Here's what worked for me:
```
diff --git a/js/msp/MSPHelper.js b/js/msp/MSPHelper.js
index 840c6fe..1167ce8 100644
--- a/js/msp/MSPHelper.js
+++ b/js/msp/MSPHelper.js
@@ -1674,8 +1674,8 @@ var mspHelper = (function (gui) {
 
         // For API > 2.0.0 we support requesting payload size - request 4KiB and let firmware decide what actual size to send
         if (CONFIG.apiVersion && semver.gte(CONFIG.apiVersion, "2.0.0")) {
-            buffer.push(lowByte(4096));
-            buffer.push(highByte(4096));
+            buffer.push(lowByte(256));
+            buffer.push(highByte(256));
         }
 
         MSP.send_message(MSPCodes.MSP_DATAFLASH_READ, buffer, false, function (response) {
```
This allowed me to download full data flash content (8MB) with average speed ~47kbps (on 115kbps link).

krzysztofmatula, 2017-10-18:
Forcing MSPv1 here (by commenting out the 2 buffer.push() lines) also works and the average transfer speed is ~32kbps.

Dronek, 2017-11-05:
Same problem here. I'm unable to download the log with the OS X configurator since I updated from iNav 1.7.3 to 1.8.

alexeystn, 2017-11-23:
I've got the same problem on my OS X Configurator 1.8.1 and iNav 1.8.

DzikuVx, 2020-05-19:
Tested with 10.15.4 and Configurator 2.5: works like expected

closed 2020-05-19 by DzikuVx