iNavFlight/inav-configurator #304: Mag hardware:  Mixed up flag bits

straferr, 2017-11-20:
This issue seems similar to #287, where the bit flags for Rx type were not read/set correctly from the configurator GUI.  This time, however, it appears to be for the magnetometer.

I am converting a old Naze32r5 acro to use an external baro and compass (HMC5883) via i2c on iNav 1.7.3 Beta.  The baro is detected fine and works as expected.  However, in order to get the firmware to recognize the mag, I have to set the mag type to "AUTO" in the gui.  This results in a fully working mag (tested in the sensors tab), but shows up as "FAKE" in the GUI, and "QMC5883" in the CLI.  I have tested two different and separate HMC5883 mag modules (the popular GY-271 and GY-273) and the GUI/CLI results are the same as described.

A second issue is that the HMC5883 is listed as QMC5883 in the CLI.  Forcing it to HMC5883 in the CLI results in a non-functioning configuration.  Very odd, considering the GUI is not involved here.

Apart from these superficial appearances, the external mag and baro work simultaneously and without other issues under iNav 1.7.3 beta if the "AUTO" gui option is used and the software/firmware are allowed to "fall where they may".   So, you still have a functional system, just a bit confusing.

straferr, 2017-11-21:
Forgot to mention above that  I was using iNav Configurator 1.8.1.  Today, I can also re-confirm this behaviour after just having added an external HMC5883 mag module via soft i2c on a brand new Matek F405OSD. ie. mag is only working properly when selecting "AUTO" in the gui, which then subsequently shows up as "FAKE" after reboot (but works fine).

DzikuVx, 2018-02-22:
This seems to be fixed in 1.9

closed 2018-02-22 by DzikuVx