iNavFlight/inav-configurator #307: quick 'no reboot' save for current sensor settings & calibration

csurf, 2017-12-06:
Currently, the current sensor settings are located within Configurator's general "Configuraton" settings tab, which has a 'save & reboot' button in order to save the settings. This causes the current calibration process to be a slower, more tedious process. IOW, if you're trying to play with the offset & scaling values for the current sensor, you're forced to wait for the board to reboot before seeing if the changes 

Can we please consider moving the current sensor settings to a different tab where we can save & apply changes to the current settings instantaneously without having to go through a reboot? This way, we can see the results of the changes in realtime. It could be via a simple 'save' button (like in the 'Modes' tab). 

Does this sound acceptable?

Arvycka, 2018-03-06:
That would be very useful - especially for a fixed wing where iNav is trying to break servos and linkages on every reboot. (What is that servo twitch about anyway?)

marianomd, 2019-10-05:
Yes please. By itself, current sensor calibration is a PITA. Requires lots of iterations. This should be copied from betaflight configurator, so it's a little easier. Also, showing the amperage in the motors tab is needed.