iNavFlight/inav-configurator #312: Linux build (gulp task)

xswein, 2017-12-18:
Hello,
will there be standalone releases for inav configurator? I cannot figure-out howto compile app, since gulp target for linux* is not provided - since betaflight configurator has linux release it should be possible and not that complicated.
Thanks
Martin

blackmennewstyle, 2018-01-08:
I'm second to that, please don't forget us, kind and long white beard GNU/Linux users :)

juanquy, 2018-02-05:
I got to run ./node_modules/gulp/bin/gulp.js release under Ubuntu 17.10 it created the folder /apps with 3 compressed files for Windows, Mac, Linux  and a folder called inav-configurator , I uncompress the Linux64 folder and tried to run ./inav-configurator file, but is not runing. Any ideas How to build the app for Linux?

wymbly1971, 2018-02-10:
For Linux please...Ubuntu/Mint

jimmycapizzi, 2018-02-11:
Don't forget us! Linux version is s must!

stronnag, 2018-02-18:
This is now in 1.9.0+.  It would be nice if:
* There was also a 32bit build (i.e. clone the 64bit gulp tasks)
* The icon was separately in the archive (to make it easy to generate a UI XDG `.desktop` file
* The path did not contain a space (c.f. the βF approach where the debian package ends up in `/opt/betaflight/betaflight-configurator/`)

Example desktop file etc: http://seyrsnys.myzen.co.uk/inav-configurator-linux/

blackmennewstyle, 2018-02-18:
Nice :)
I actually did the same kind of install for Firefox Quantum on my GNU/Linux Debian.

Thank you so much guys. We love you!

welon68, 2018-02-18:
Thank you for linux versions !!!

closed 2018-02-23 by xswein

MarekSpy, 2019-03-02:
Hi
I tried to run inav configurator but it does not work. I did according to the way as on this page  http://seyrsnys.myzen.co.uk/inav-configurator-linux/ 
But the program has some error and I can not cope. I have to use version 1.9.4 because I have cc3d Atom. I do not know how to run it

stronnag, 2019-03-02:
This version is not supported by the iNav project.

However:
* It works for me
* As you've neglected to describe exactly how the application fails for you, there is no way to help you.

MarekSpy, 2019-03-02:
Hi

I did it differently and it works. Yhx for help