iNavFlight/inav-configurator #330: Chrome Configruator on Linux - Can't Calibrate

linuxslate, 2018-01-23:
Chrome Configruator 1.8.1 On Ubuntu 16.04

It only gives me about 2-3 seconds to calibrate.  It is impossible to complete.

I connected the FC (Omnibus F4 V3 Pro) to Android EZGUI, and was able to calibrate with no problem.

```
2018-01-22 @ 22:18:21 -- Accelerometer calibration started

2018-01-22 @ 22:18:23 -- Accelerometer calibration finished

2018-01-22 @ 22:18:27 -- Accelerometer calibration started

2018-01-22 @ 22:18:29 -- Accelerometer calibration finished
```

DzikuVx, 2018-01-23:
This is how it's supposed to be working: https://quadmeup.com/inav-advanced-accelerometer-calibration-why-and-how/

closed 2018-01-23 by DzikuVx

linuxslate, 2018-01-23:
Why is this on some un-related website.  I did read the inav documentation on the inav wiki, and I searched but did not find this.

There should be a step counter, or something that guides the user through the steps. EZGUI does it very nicely.  Even just "Accelerometer calibration step X finished " would be adequate.

I guess "Accelerometer calibration step X of 6 finished" is asking way too much.

Close this if you want, but it still looks like something is missing from the configurator to me.

DzikuVx, 2018-01-23:
What you are referring to right now is not what you reported in a first place. 
And that is already fixed and will be available in next release

digitalentity, 2018-01-23:
@linuxslate 
1. Procedure is also documented in the INAV Wiki here https://github.com/iNavFlight/inav/wiki/Sensor-calibration#accelerometer-calibration-steps
2. Upcoming Configurator/firmware release has the better UI for sensor calibration

linuxslate, 2018-01-23:
Summary of this issue thread:

1.  There is nothing wrong with the code.  It was just due to a dumb user that didn't read (or understand) the documentation.

2. We've already fixed the problem in the code.

... And Developers wonder why users complain.  :-/

skaman82, 2018-01-23:
This is a RTFM problem and waste of developer time. So no reason to complain about anything