iNavFlight/inav-configurator #335: My Setup is "well working" with BF - but not with INAV

artusmuc, 2018-01-31:
My setup:
QUAD-X 
HGLRC F4 V5 (Omnibus F4 target)
MS5611 baro
GY-262 MAG 
Frsky X4RSB (Walkera F12E with deviation 5.0)
2 S lipo
DYS-12A Blheli-S ESCs multishot 142
1104-8700 kV motors (Ok i know the this are Chinese KV)
Motor timing middle-high 

My question is - with BF flashed (stardard parameters)  nothing changed - the first test - hover and fly - and mostly do what i want.

INAV - hovering - wobbling - shoot in the air - crash - result of more than one test is: unflyable - i even replaced FC - same result. Without props motor testing - with moving the quad seems all ok. After mounting probs - with a little throttle - extremely wobbling and out of control - motors not spinning down with throttle down. 

I have tried in RCgroups - but (except of stronnag) nobody answers

digitalentity, 2018-02-01:
Betaflight is optimized for miniquads, while INAV does not. Try applying a 5" racing preset and check if it flies.

artusmuc, 2018-02-02:
Thank you for helping me - i havn't used any preset - i will try this on Sunday. As i understand your response any other FC will react with INAV not better - i have the Matek OSD also. 
My plan is first to build a small quad - with this test all what i want to have on quad cause the possible damage of a small quad is more acceptable than with 20 cm - or a 650 with 4110 motors (i have all "on stock" :)
Can i try to screw on some parameters to make INAV a possible working with a microquad? 

My first test with INAV was ok - but after i activated MAG - this quad start with extrem wobbling after mounting probs and hovering. 

Thank you in advance for your help.

artusmuc, 2018-02-04:
Hello Konstanin,

I have read some "how to's" and some threads.  So i flash INAV now again and use "standard" presets only changing Gyro LPF cutoff to 120 Hz - and without Baro & Mag the little bird is proper hovering and flying stable. Only Landing is now a problem - touching ground and then it flips and is out of control. 

So i think INAV is fine for micro quads - when the user has a little bit more skill than me :) 

I don't know - but i think there are a lot of users with small quads cause 90 to 110 mm frames are cheap and available - also the 20 mm  F4 FC's 

maybe the inav user can have a preset for a small quad in future :) 

Playing with "PID tuning"  i have planed to learn in future - but i have to learn now abut this parameters. 

Thanks for your kind help !

artusmuc, 2018-02-05:
With a second trial "landing" is now proper possible - increasing ACC cutoff to 30 Hz. Thanks a lot ! I am a little bit (more) happy now.

closed 2018-02-05 by artusmuc