iNavFlight/inav-configurator #351: Check if version string is received before checking firmware family and version

DzikuVx, 2018-02-19:
>  I've stumbled upon an interesting issue. Once in a while Configurator will connect to CLI as if firmware family is mismatched. Doesn't happen every time and after reconnecting it's fine. I wonder if we should ensure that version string is received from FC before checking firmware family and version in Configurator.

digitalentity, 2018-02-19:
Possibly https://github.com/iNavFlight/inav-configurator/issues/350 was a symptom of the same bug.

DzikuVx, 2018-02-19:
That is very probable. To be honest I have been experiencing this one or twice before and it indeed looks like a race condition

Tygrys-1, 2018-05-08:
Got this one today with configurator 1.9.3, iNav 1.9.1 on Omnibus F3:

2018-05-08 @ 10:37:10 -- Serial port successfully opened with ID: 5

2018-05-08 @ 10:37:10 -- MultiWii API version received - 2.1.0

2018-05-08 @ 10:37:10 -- Flight controller info, identifier: INAV, version: 1.9.1

2018-05-08 @ 10:37:10 -- Running firmware released on: Apr 21 2018 12:53:03

2018-05-08 @ 10:37:10 -- Board: OMNI, version: 0

2018-05-08 @ 10:37:10 -- Unique device ID received - 0x2300184634570c20313835

2018-05-08 @ 10:37:31 -- EEPROM saved

2018-05-08 @ 10:37:31 -- Device - Rebooting

2018-05-08 @ 10:37:31 -- Serial port successfully closed

2018-05-08 @ 10:37:35 -- Serial port successfully opened with ID: 6

2018-05-08 @ 10:37:35 -- MultiWii API version received - 2.1.0

2018-05-08 @ 10:37:36 -- Serial port successfully closed

2018-05-08 @ 10:37:36 -- You need to upgrade your firmware before you can use the Configuration tab.
[image]

shellixyz, 2019-03-24:
@DzikuVx Can't this be closed ?

DzikuVx, 2019-03-25:
I did nothing with this. And it's still poping up from time to time... Let's leave it open, maybe I will dif into it a little

closed 2019-11-20 by DzikuVx