iNavFlight/inav-configurator #360: Show navPos[2] value

giacomo892, 2018-02-25:
Could be neat is we can display the navPos[2] value in meters inside the brackets so it can be directly compared with baro. 
[image]

DzikuVx, 2018-02-25:
@giacomo892 wrong project man :)

giacomo892, 2018-02-25:
@DzikuVx hahahahah I'm deeply sorry man LOL I opened the issue 15m after I left the bed. Should I close and open there?

DzikuVx, 2018-02-25:
Yes please

Paweł Spychalski

Od: giacomo892
Wysłano: niedziela, 25 lutego 2018 20:08
Do: iNavFlight/inav-configurator
DW: Paweł Spychalski; Mention
Temat: Re: [iNavFlight/inav-configurator] Show navPos[2] value (#360)

@DzikuVx hahahahah I'm deeply sorry man LOL I opened the issue 15m after I left the bed. Should I close and open there?

giacomo892, 2018-02-26:
Done. Closing.

closed 2018-02-26 by giacomo892