iNavFlight/inav-configurator #373: Issues with RTH

Deidfly, 2018-03-03:
Hello I have an issue with the RTH of my Nano Talon:

https://www.youtube.com/watch?v=4Vkr9FsFnVo

After a positive maiden flight (exept heavy wingrock) the plane came back home and did everything I expected. Suddenly the plane doesn´t come home again and I have no Idea why. As you can see from the footage the plane turns correctly 180 degrees but than instead of moving towards the homepoint it starts to circle 8.
My setup: 
Omnibus F4 V2.1 INAV 1.8
Neo M8N
L9R Receiver
20A ESC with 3A BEC on 4s 1300mAh Lipo
600mW 5.8GHz video Transmitter

shellixyz, 2018-03-03:
Please post a dump of your settings and also blackbox flight log if possible.

shellixyz, 2018-03-03:
It looks like your plane is stalling. Try to increase the cruise throttle to 1620.

`set nav_fw_cruise_throttle = 1620`

digitalentity, 2018-03-04:
@Deidfly please post this issue in firmware bug tracker https://github.com/iNavFlight/inav/issues
Please provide full settings dump there so we could verify the settings.

closed 2018-03-04 by digitalentity