iNavFlight/inav-configurator #395: 1.9.2 configurator launches minimized

artempolikarpov, 2018-03-25:
OS: Windows 7x64 SP 1

Newly unzipped 1.9.2 configurator launches minimized.
It is displayed in task bar and processes list, but minimized and it is impossible to maximize. 

Restarts don't help.

Tried different approaches: as admin and not, also compatibility options for Win 7 - but nothing worked for me.

Afinogen, 2018-03-30:
@artempolikarpov Delete config folder in AppData

Нужно удалить конфигурационный файл в папке AppData, вроде как фиксил это...

artempolikarpov, 2018-04-01:
There is no config file or folder inside AppData or iNav folder

Afinogen, 2018-04-01:
@artempolikarpov look in  C:\Users\%Your_UserNname%\AppData\Local\inav-configurator

artempolikarpov, 2018-04-01:
There is only User Data folder

Afinogen, 2018-04-02:
@artempolikarpov remove  C:\Users\%Your_UserNname%\AppData\Local\inav-configurator\User Data folder. After starting the configurator, it is automatically created

RAMukOK, 2018-04-07:
I was having this issue.  Deleting the user data folder fixed it.

BTW, you will have to go to tools folder options and check view hidden files folders and drives to see the folder in question.

closed 2018-05-13 by DzikuVx

craig34a, 2018-06-18:
Thank you. This is an issue after reinstall in a different folder.