iNavFlight/inav-configurator #398: big big problem.....

Christianmix, 2018-03-29:
Hi everyone, I have 2 problems, the first is in relation to the curve in angle mode, the model makes a very open curve even with every command for the squad, it goes very far to get back, because due to this problem I flew in air mode and flew well, but as it appears in the video, the altitude is varying a lot, it is in meters, see how it varies, is there anything I can do, happened with 2 omnibus f3 boards today, it happened to mine flying wing and with my sky walker, the altitude is crazy, both are with the same version of the inav, but only in my flying wing I had problem as the curve very open, follow along the dump

https://youtu.be/KH9JzVS8fwo

# version
# INAV/OMNIBUS 1.9.0 Mar  4 2018 / 23:17:26 

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 -100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature MOTOR_STOP
feature GPS
feature CURRENT_METER
feature RSSI_ADC
feature BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD
feature FW_LAUNCH

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
[... 501 lines cut ...]
set nav_mc_vel_xy_p = 40
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 50
set thr_expo = 0
set tpa_rate = 33
set tpa_breakpoint = 1300
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 37
set pitch_rate = 11
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

Christianmix, 2018-03-30:
ok, I discovered part of my problem, the part of the long curves .. :)
set max_angle_inclination_rll = 300
set max_angle_inclination_pit = 300
Now it remains to know what happens to the altitude........

Christianmix, 2018-03-30:
I was looking here and found about putting the weight of gps at 0 for altitude measurement so I left it like this: set inav_w_z_gps_p = 0 and set inav_w_z_gps_v = 0 

is it right?

shellixyz, 2018-04-07:
Please read the comments in [this issue](https://github.com/iNavFlight/inav/issues/2506) and the ALTHOLD documentation in the wiki

closed 2018-05-03 by DzikuVx