iNavFlight/inav-configurator #399: Roll and Pitch output are swapped in stabilized mode

craig34a, 2018-04-01:
Passthrough mode produces correct control outputs. 
Stabilized mode swaps the roll and pitch responses. 
When the aircraft is moved on the pitch axis, a roll response follows and 
when it is moved in the roll axis, a pitch response follows. 
How to swap these? 

Flying wing, Omnibus F4 Pro V3, Taranis, L9R, iNav 1.9.

INAV/OMNIBUSF4V3 1.9.0 Mar  5 2018 / 00:18:55 (dc4ef594f)

Entering CLI Mode, type 'exit' to return, or 'help'

# smix
smix 0 3 0 50 0
smix 1 3 1 50 0
smix 2 4 0 -50 0
smix 3 4 1 50 0

# dump

# version
# INAV/OMNIBUSF4V3 1.9.0 Mar  5 2018 / 00:18:55 (dc4ef594f)

# resources

# mixer
mixer FLYING_WING

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0
smix 1 3 1 50 0
smix 2 4 0 -50 0
smix 3 4 1 50 0

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 840 2000 1340 -80 -1 
servo 4 1000 2160 1700 -80 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
[... 524 lines cut ...]
set nav_mc_vel_xy_p = 40
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 46
set thr_expo = 0
set tpa_rate = 33
set tpa_breakpoint = 1300
set rc_expo = 67
set rc_yaw_expo = 20
set roll_rate = 32
set pitch_rate = 9
set yaw_rate = 19
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

digitalentity, 2018-04-02:
Definitely setup issue. See https://github.com/iNavFlight/inav/issues/2320#issuecomment-375935020

DzikuVx, 2018-04-03:
One of the servos should be reversed while the other not

closed 2018-04-03 by DzikuVx